plasma-sds/renate #1: Create repository structure

gergopokol, 2017-02-24:
Repository structure is to be created hosting a folder for documentation and a module for rate_equation_solver.

gergopokol, 2017-02-24:
Preparations done.

gergopokol, 2017-02-24:
Some changes needed...

gergopokol, 2017-03-10:
Strategy is the following: The repo is organized into **folders that are independently deployable packages**. Each package contains **modules that typically contain one class each**. For each class we usually also have a test module. For more general information on the organization of the development process, see [**Programming guidelines**](https://github.com/gergopokol/renate-od/blob/master/programming_guidelines.pdf).

gergopokol, 2017-03-30:
Need new .gitignore

gergopokol, 2018-03-02:
Create and maintain **conda** configuration. Collect best practices, and decide the way to do it.

SzondyB, 2018-03-21:
On conda setup:
https://conda.io/docs/user-guide/tasks/manage-environments.html#creating-an-environment-from-an-environment-yml-file

gergopokol, 2018-03-30:
**Renamed setup file to be more readable!** --> Should accept theirs when merging.

Creates conda all right, yet there are still some package errors:
`
from crm_solver.beamlet import Beamlet

Backend Qt5Agg is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "C:\Users\Pokol\AppData\Local\conda\conda\envs\renate_od_conda\lib\site-packages\IPython\core\interactiveshell.py", line 2910, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-94e6a3e1282c>", line 1, in <module>
    from crm_solver.beamlet import Beamlet
  File "C:\Program Files\JetBrains\PyCharm 2017.2.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "A:\9github\renate-od\crm_solver\beamlet.py", line 6, in <module>
    import matplotlib.pyplot
`

gergopokol, 2018-03-30:
One can **create the conda** by running: `conda env create -f conda_setup.yml`
and update by running: `conda env update -f conda_setup.yml`

Then one must set the interpreter of the project to the python.exe in the renate_od_conda directory.

gergopokol, 2019-06-19:
The only living task here was to find out the situation with Conda. For the time being, we consider Conda to be unnecessary. We can handle compatibility issues by specifying Travis tests and making a setup script #46

closed 2019-06-19 by gergopokol