plasma-sds/renate #2: Implement constant parameter ODE solver

gergopokol, 2017-02-28:
Rate equations are a set of ordinary first order linear differential equations. For constant parameters it is possible to solve this analytically, offering an ideal test case. As a first step, the constant parameter version of this solver is to be implemented along with its unit tests.

gergopokol, 2017-04-11:
@SzondyB :
TestOde should be refactored following example of test_1d_analytical, and test suit should be extended for more coverage.

gergopokol, 2017-09-07:
`test_set_up_equation` has ben ammended with two dummy parameters to run, but I am not at all sure why we need inputs `calculation_point` and `steps` in s`et_up_equation(variable_vector, calculation_point, coefficient_matrix, steps)` is needed. Without is, however, the `odeint` solver does not run.

asztalosors, 2017-10-20:
The input data object allocation is faulty. If different data sources are to populate the object ex: constant plasma, hdf5 plasma input, ids input, an intelligent switch has  to be implemented to avoid rewriting the code.

asztalosors, 2017-10-20:
The routine has a bug, when running "set_up_equation" in line 39 of ode.py.
"derivative_vector = numpy.dot(variable_vector, interp_coefficient_matrix(actual_position))"

The error is related to the usage of the interp_coefficient_matrix function outside of the interpolation range allotted.

asztalosors, 2017-10-20:
The temporary bug fix implemented in 36d9ef4 has to be further attended to and the actual interpolation range overflow fixed.

asztalosors, 2017-10-20:
Preliminary beam evolution results for constant plasma
[image]
.

asztalosors, 2017-10-20:
The routines suffer from object duplication, input allocation is manually resolved, overflow errors.

gergopokol, 2018-12-04:
Constant parameter solver  is to be removed, because it there only for historical reasons.

closed 2018-12-04 by gergopokol