plasma-sds/renate #3: Implement rate loader routine

gergopokol, 2017-02-28:
Rates (and any other data) should not be stored in the source code repository, rather it should be stored locally, and downloaded from a website if needed.

plasmatyi, 2017-05-08:
Dear Gergo Pokol,

The email sent to github@maradi.tk with subject
"Re: [gergopokol/renate-od] Implement rate loader routine (#3)"
will be delivered as soon as possible.

Your email address is automatically added to the
recipients 'whitelist'. This means for further
emails to github@maradi.tk, you will not receive a delivery
notification again.

TK MAILIAS
*************************************************
Looking for a way to Protect Your Identity and 
for a great no-more-spam solution for your email
address reply@reply.github.com? Sign up for TK Mailias.

It's free! Please visit http://www.mail.tk/

asztalosors, 2017-09-08:
Correction made to routine regarding relative paths. The correction needs to be test in different environments.

plasmatyi, 2017-09-08:
Dear asztalosors,

The email sent to github@maradi.tk with subject
"Re: [gergopokol/renate-od] Implement rate loader routine (#3)"
will be delivered as soon as possible.

Your email address is automatically added to the
recipients 'whitelist'. This means for further
emails to github@maradi.tk, you will not receive a delivery
notification again.

TK MAILIAS
*************************************************
Looking for a way to Protect Your Identity and 
for a great no-more-spam solution for your email
address reply@reply.github.com? Sign up for TK Mailias.

It's free! Please visit http://www.mail.tk/

gergopokol, 2018-02-23:
General purpose utility development continued in #13 . Specialized rate loader continued here.

gergopokol, 2018-03-02:
It could be considered to move this routine into the general purpose GetData class. Testing and review is necessary.

gergopokol, 2018-03-09:
All functions moved to GetData in #15.

closed 2018-03-09 by gergopokol