plasma-sds/renate #5: Implement changing parameter ode solver

SzondyB, 2017-03-16:
Extension of constant parameter ode solver.

gergopokol, 2018-03-09:
**Major restructuring** in the `feature_solver-refactoring` branch.

gergopokol, 2018-03-10:
**CRM solver works** with automatic data download and XML input files. Should be further tested, and further refactoring is to be done with continuous regression testing. Can be run from Console:

```
from crm_solver.beamlet import Beamlet
b = Beamlet()
b.plot_populations()
```

gergopokol, 2018-03-12:
This is to be implemented! :)

[image]

gergopokol, 2018-03-20:
This is technically working, so merged to master.
The result is, however, clearly wrong, with unchanged level0. The problem is most probably in `CoefficientMatrix.assemble_terms()` with mixing up the dimensions of the rate matrixes.
This is also probably not ok: `from_level + 1:self.number_of_levels`. 
This needs to be checked and benchmarked to RENATE!

gergopokol, 2018-03-21:
test_profiles.h5 density was in wrong dimensions. Had to be multiplied by 10^19. The structure also changed by introducing the `'profiles'` key to test_profiles.h5. **Needs to be re-downloaded!**

Output looks nice. Should benchmark to old RENATE!

SzondyB, 2018-04-05:
Successful Benchmark:
[image]

gergopokol, 2019-06-20:
This is being continued in #56 and #50 .

closed 2019-06-20 by gergopokol