plasma-sds/renate #7: Visualization of population evolution along beam

gergopokol, 2017-05-11:
Output of the 1D beam simulation is to be visualized from saved hdf5 output. All ipnut data and population evolution is to be plotted.

SzondyB, 2018-04-18:
To do:
- self.profiles read to __init__
- setting axis properties to separate functions
- plotting functions: create axis, call function

gergopokol, 2018-06-26:
access_path should point to the xml file specifying the run parameters. The path to the corresponding h5 file is to be read from it. Similarly to the Beamlet class.

gergopokol, 2018-06-26:
benchmark_to_renate() is to be changed to benchmark(reference_data_path) routine that can compare any arbitrary profile result files. 

Run parameters are also to be plotted and compared. These should also be plotted on the plot_all_profiles() plots.

gergopokol, 2018-09-05:
xml files should be taken as input, and information there to be visualized on the plot.

asztalosors, 2019-06-19:
The issue has long been fixed.

closed 2019-06-19 by asztalosors