plasma-sds/renate #8: 1D observation module

asztalosors, 2017-05-11:
Create an 1D observation module.
- emission along beam calculated by multiplying with beam current and spontaneous emission fractions.
- photon current integration along beamlet for designated sections along the beam.

gergopokol, 2018-08-08:
Beamlet current is to be added as property of the Beamlet class, and read from there.

gergopokol, 2018-08-08:
Renames necessary:
input_obj --> beamlet
input_param  --> beamlet_param
input_profiles --> beamlet_profiles

gergopokol, 2018-08-08:
calculate_light_profile() to be refactored using vector operations and short internal variable names

gergopokol, 2018-08-08:
self.input_param = input_obj.param
        self.input_profiles = input_obj.profiles
can be removed, and use references like beamlet.param instead.

asztalosors, 2019-04-16:
Functionality of this module will be taken over by CHERAB. Do we need this issue alive?

SzondyB, 2019-09-03:
[image]
Observed photon emission profile: with - blue - and without interpolation - orange.

closed 2019-09-03 by SzondyB