plasma-sds/renate #10: Read from IMAS

gergopokol, 2017-10-20:
Routines to read from IMAS database and save data on other formats.

asztalosors, 2018-03-09:
Discussion on how IMAS relates to Conda. 
In case of a python actor can the interpreter be specified?
What would be the default procedure of the installation of python routines be?

gergopokol, 2018-03-20:
Opened a dedicated branch: feature_IMAS

asztalosors, 2018-03-22:
Current init routines need testing before implementing detailed getter routines.
To be done on iter cluster.

asztalosors, 2018-03-22:
init of the IDS take in any machine name as long as user is not defined. Please correct.
Otherwise the routines load IDS-s from local data base.

Criteria of local data base access is to:
1. Create it: imasdb iter
2. use command: scenario_summary to list all IDS-s
3. Copy the one you need or want from /work/imas/shared/iterdb/3/0 to your local db

asztalosors, 2018-03-22:
Global IDS access is shoddy right now. Needs debugging. Not a priority @ the moment.

asztalosors, 2018-03-22:
Loading sequence for CoreprofIds and EquilibriumIds do not work. Inheritance not implemented correctly.

asztalosors, 2018-03-22:
Inheritance Issues are resolved. The routines load the speciffic IDS-s. Time to work on attributes for data retrial.

asztalosors, 2018-03-23:
Currently implemented all the features needed from the core_profile IDS. Further upgrades to be added when requested.

asztalosors, 2018-03-23:
Ok. So it turns out most of the IDS-s don't have rho_tor_norm, neither psi. The equilibrium file has psi. So further grid implementation extraction needed.

SzondyB, 2018-04-10:
ImasObject doesn't work for me unless I give my username and device name - maybe a default value should be added istead of 'if value is None'.
It can't download data from the iter work directory, only from local user directory if the file has been copied there before.

gergopokol, 2018-06-26:
File names should be consistent with the class names!

asztalosors, 2018-10-05:
The data used by renate-od from core_profiles and edge_profiles is located in identical data structures. It makes no sense to create a separate classes of core and edge profiles. Edge profiles can and should inherit the functionality of core_profiles or a universal class called profiles should be instated instead, where a property of the class defines the data source.

gergopokol, 2019-09-04:
The decision on abandoning the whole package postponed until CHERAB-IMAS integration is finished.