plasma-sds/renate #11: IMAS wrapper for basic BES model

gergopokol, 2017-10-20:
The wrapper should read from IMAS database, and run basic RENATE-OD simulation

asztalosors, 2018-04-10:
Note to all users: The wrapper_1d is set up to read any coreprof data, it creates an element tree form imas data and appends to it additional data that the solver needs like beam energy, beam current and species which will be grabbed from IDS-s currently the grabbing routines are hard coded but adequate error handling has been input for easy improvements.

asztalosors, 2018-04-10:
Note to all users: The wrapper is hard coded to operate with the assumption that the beam length is 10 cm. The n(rho) where rho in range [0 -> 1] is density/temperature functions are down-scaled on the 10 cm of beam.

This HAS TO BE revised but for WPCD CC @ Poznan 2018 IDS testing it will suffice.

asztalosors, 2018-04-10:
Code is written as such that refactoring and extension should come easy to it.
Not tested yet.

asztalosors, 2018-04-10:
1D wrapper tested and functions successfully. Needs a benchmark with RENATE :D

gergopokol, 2018-06-26:
get_beamlet_... routines are to be implemented in the imas_utility package in the NbiIds class. Similarly to the CoreprofIds class.

asztalosors, 2018-10-16:
The profile allocation is faulty. The following process should be implemented: flux = 0 --> 1 : core profiles edge profiles if there is. Extrapolate using exponential function if edge profiles not available. Exponential approach to be applied as well for beam or flux values out of bound.

gergopokol, 2019-09-04:
The decision on abandoning the whole package postponed until CHERAB-IMAS integration is finished.