plasma-sds/renate #15: Refactoring and unit testing of utility routines

gergopokol, 2018-03-02:
`GetData` and `Settings` utility classes should be refactored and unit tests implemented. These are general purpose features to be used in other projects, as well, so particular care should be taken.

gergopokol, 2018-03-02:
Might need infrastructure to read from XML config files produced by the measurement systems. Some examples added.

[APD-XML.zip](https://github.com/gergopokol/renate-od/files/1775035/APD-XML.zip)

gergopokol, 2018-03-06:
Downloaded **data version should be logged** and made available for routines that in turn should include it in the output. Version numbers are to be stored in the XML records in the data root folder. The `dummy `property should also be stored in this file. A refresh switch is to be implemented in the data reader, which would download the most recent version of the data files requested. 

On the long run, data will be stored in an SVN repo that is to be checked out to the appropriate folders of the data user. Data downloader will have to access the SVN revision number for the selected file, and store it the local data XML.

gergopokol, 2018-03-09:
Use of https://github.com/martinblech/xmltodict might be considered but only after solving the environment is set up in #1.

gergopokol, 2018-03-10:
beamlet/test.xml corrected (<meta> tag had no closing --> deleted) xml reader in GetData tested.

gergopokol, 2018-03-10:
new version of beamlet/test.xml uploaded

gergopokol, 2018-03-10:
GetData tested for XML reading.
GetData buggy for H5 reading. --> Need debugging!

gergopokol, 2018-03-10:
Using HDFView+Object 2.14 https://support.hdfgroup.org/products/java/release/download.html to check h5 file.

gergopokol, 2018-03-10:
beamlet/test_profiles.h5: typos corrected, converted using convert.py.

gergopokol, 2018-03-21:
beamlet test data files had to be updated at the public dummy directory. **Public rates data still missing**, so now there is no way to run the code on dummy data.

gergopokol, 2018-03-21:
SCP download fails on some Windows computers due to "'pscp' is not recognized as an internal or external command, operable program or batch file." This might have to do with setting msi installer (https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) to add path to cmd path. To be tested.

gergopokol, 2018-03-21:
**'pscp' is not recognized**:  Problem was indeed solved by properly installing putty.

gergopokol, 2018-03-30:
One does not need access to the data user of the deep server to access the private data. It is enough to have any access to the deep server, and set the getdata_setup.xml as:

```
    <server_private_address>user@deep.reak.bme.hu:/home/data/private_html/renate-od</server_private_address>
    <private_key>full_path_of_key_file.ppk</private_key>

```

gergopokol, 2018-08-06:
Beamlet.write_beamlet_profiles() is to be moved to utility.writedata.write_beamlet_data() with inputs of param and profiles variables and location data read from getdata_setup.xml.

This routine is then to be used to save both beamlet calculation results and data read from IMAS in wrapper_1d.

gergopokol, 2018-08-08:
getdata.setup_rate_coeff_arrays() and getdata.get_mass() should not be in the general purpose utility package.

gergopokol, 2018-08-08:
getdata.get_mass()  can be moved to Beamlet class, and at the initialization of Beamlet mass can be loaded and added to the param tree along with the calculated velocity.

getdata.setup_rate_coeff_arrays() can be moved back to the original location in Rates class, and when calling initialize_ode() of the Beamlet class, the Einstein coefficient(s) responsible for the observed line can be added to the params tree. Observed transition is to be added to the input params structure as optional parameter. If given, the given transition is to be used, if not, the default should be set based on the beam species.

gergopokol, 2018-08-08:
Example of xmlTree manipulation: #12

asztalosors, 2019-09-24:
Removed redundant code from utility package. The code is ready for unittest development and subsequent optimization. May the fun begin :)

gergopokol, 2020-09-28:
This has been mostly done. Follow up issues: #136 #137 #110 #139 #140

closed 2020-09-28 by gergopokol