plasma-sds/renate #17: Device module

asztalosors, 2018-04-10:
IMAS development prompted the need for a device module.
This needs to be a set of universal objects and functions that handle device specific magnetic geometries, observation geometries etc. The routines have to be universally applicable to imas and in the stand alone package as well. Name of module is still debatable.

gergopokol, 2018-08-06:
Related wiki page: https://github.com/gergopokol/renate-od/wiki/Specfication

asztalosors, 2018-10-15:
Device module handling requires routines that calculate:
1. distance between 2 points
2. unit_vetor between two points
3. coordinate transformation from cylin to cart and back

asztalosors, 2018-10-15:
For above mentioned issue the following package might be explored:
https://docs.sympy.org/latest/modules/geometry/index.html

For now the requested routines will be added to the convert within the utility package

asztalosors, 2019-06-19:
This issue should be closed and continued under CHERAB within the proper CHERAB device modules for beam and observation generation.

gergopokol, 2019-09-04:
The decision on abandoning the whole package postponed until CHERAB integration is finished.

asztalosors, 2021-06-23:
The device module requires a somewhat redefinition of what it may contain. 
The requirements for the device module entail the following:
* Plasma
* Beam
* Equilibrium
* Observation geometry