plasma-sds/renate #21: Introduce clean coding

mindler8, 2018-10-18:


mindler8, 2018-10-18:
Our purpose to add some useful clean coding convetion to this branch.

mindler8, 2018-10-19:
1. change: Constants are in uppercase, eq: MY_CONSTANT = ...

mindler8, 2018-10-19:
Note: We follow the PEP8 conventions: https://www.python.org/dev/peps/pep-0008/

mindler8, 2018-10-20:
MISSING: value comparision from **test_size_of_set_up_equation_constant** function

asztalosors, 2018-12-03:
The `test_setup_derivative_vector_with_coefficient_matrix` and `test_calculate_numerical_solution` should be renamed as they test diagonal differential equation system, which is a trivial case of the equation system.

A non diagonal test should be set-up as well.

asztalosors, 2018-12-04:
1. ode.py to be cleaned up: Only 3D coefficient matrices are to be considered in function `setup_derivative_vector`.
2. Test to be implemented:
    2.1. Type, size, dimension and functionality tests for all existing functions within.
    2.2. Test cases: constant diagonal case; constant non-diagonal case; non-constant non-diagonal case.
    2.3. Tests should be dynamic: a formula function to be included that computes the expected results (should allow for changing up the input test cases if need arises without having to redo the manual calculations)
    2.4. Error of the method should be tested as well.

asztalosors, 2018-12-04:
If unit testing fails for any reason a `.cache` folder is being created within the module in question.
This folder should be added to the git-ignore list.

asztalosors, 2018-12-07:
Should the `test_two_solutions_with_each_other` test case be removed? Two different solvers are compared, though their agreement is recommended, it should not be expected. I'd prefer a this step to occur within a benchmark.

asztalosors, 2018-12-07:
The name of the test should include the word: benchmark and the coefficient matrix case the benchmark occurs for.

asztalosors, 2018-12-07:
The nondiagonal varying case test is still in progress

gergopokol, 2019-04-01:
Even follow-up issue has been closed, so it is high time we closed this one.

closed 2019-04-01 by gergopokol