plasma-sds/renate #26: Adding impurities

asztalosors, 2018-12-17:
The collisional radiave model (CRM) needs to be extended to accommodate the inclusion of impurities in the electron population evolution calculation. The task requires following expansion to be made:

1. Read impurity profiles from local directory, stored in h5
2. Read impurity profiles from IDS (itercluster & gateway)
3. Handle Z-eff profiles if the upper 2 are not given
4. Interpolate the impurity n and T values along the beam, based on the read profiles
5. Add the proper impurity rate to the coefficient matrix
6. Check for quasi-neutrality

asztalosors, 2018-12-17:
Implementation of quasi-neutrality checks and conformation has two approaches:
1. Deriving n_e in function of n_i and sum(n_imp).
2. Deriving n_i in function of n_e and sum(n_imp).

The latter approach should be default and preferred.

SzondyB, 2018-12-20:
Beamlet tested, runs.

asztalosors, 2019-04-01:
Merging development branch into adding-impurity

gergopokol, 2019-04-01:
Merge done. I close issue, but the code could use some refactoring in the future.

closed 2019-04-01 by gergopokol