plasma-sds/renate #33: What is the recommended installation pattern for Renata-OD?

mattngc, 2019-03-08:
Hi,

Just preparing to play with your module after some discussions with Ors at JET this week. I was wondering what the recommended installation pattern is? I didn't see a setup.py in the root folder so I'm guessing you aren't using "python setup.py install" or something similar? Maybe you are putting the root project folder on your python path manually? 

Regards, Matt

asztalosors, 2019-03-19:
Dear Matt, 
RENATE-OD has no install protocol so far.  The wikipage "How to run" contains instructions on how to operate the code. A tested and operational version will be made available to you next week.

mattngc, 2019-03-21:
Thanks Ors, no worries. Let me know when you have the version ready for testing.

mattngc, 2019-04-02:
Resolved with the getting started guide that has been added to the wiki.

closed 2019-04-02 by mattngc