plasma-sds/renate #46: Setup script

gergopokol, 2019-04-09:
Matt has proposed to add a Python setup script in pull request #39. This has been merged to branch https://github.com/gergopokol/renate-od/tree/add_setup for further exploration.

gergopokol, 2019-04-09:
Some useful info:
https://github.com/pypa/sampleproject/blob/master/setup.py 
https://packaging.python.org/guides/distributing-packages-using-setuptools/#setup-py

mattngc, 2019-04-09:
You can also have a look at the Raysect and CHERAB ones as examples:
https://github.com/raysect/source/blob/master/setup.py
https://github.com/cherab/core/blob/master/setup.py

Ours has extra stuff in it for Cython compilation but gives you an idea of what is possible.

gergopokol, 2019-06-20:
Branch add_setup deleted. This is to be continued in branch https://github.com/gergopokol/renate-od/tree/cherab_integration_pull. Having proper setup.py is actually required by CHERAB, so it is to be fixed.

gergopokol, 2019-06-26:
Tested on ITER cluster.

setup() uses Cython, so adaption to standalone RENATE-OD might decrease portability, so this is postponed.

closed 2019-06-26 by gergopokol