plasma-sds/renate #50: Benchmark to RENATE

SzondyB, 2019-04-16:


asztalosors, 2019-06-14:
Added visualization tools for benchmark. Needs some cosmetics done as well.
Benchmark Fails!
Three test-cases are highlighted:
1. Li beam in D plasma @ 60 keV (9 levels)
2. Na beam in H plasma @ 60 keV (8 levels)
3. H beam in He plasma @ 60 keV (6 levels)

There seems to be some systematic discrepancies between RENATE and ROD: 
1. ROD rates result in stronger beam attenuation and greater relative population of excited states.
2. RENATE profiles seem to be a bit smoother than those of ROD.

Solutions and investigations:
1. Double check RENATE and ROD input data. 
2. Have a closer look at previous benchmark attempt to different numerical solver.
3. (MOST likely) Recheck CRM and see whether the matrix is correctly constructed (There have been recorded issues  with the new CRM we are using).

[image]
[image]
[image]

asztalosors, 2019-08-27:
There seems to be an error in the calculation of the beam velocity. Beam energy is given in keV but for beam velocity calculation the value is not transformed in eV. Resulting in beam velocity of ~ 30 times lower then reality. This results in an overestimation of the spontaneous emission terms in the CRM. This might result in the discrepancy witnessed so far.

asztalosors, 2019-08-27:
Adding plots for regression testing of:
1. He plasma with H beam @ 60keV
[image]
2. D plasma with Li beam @ 60keV
[image]
3. H plasma with Na beam @ 60keV
[image]

asztalosors, 2019-08-27:
There are some minor discrepancies between the two plots. Under investigation. The major problem seems to be fixed in the CRM applied in RENATE-OD

SzondyB, 2019-08-27:
Juhéj!!!!!

Asztalos Örs <notifications@github.com> ezt írta (időpont: 2019. aug. 27.,
K, 10:18):

asztalosors, 2019-09-03:
Plot of relative error for Li in D plasma.
[image]

asztalosors, 2019-09-03:
Further plots of relative error: for
Na:
[image]
and for H:
[image]

asztalosors, 2019-09-03:
It is clear that the relative error on higher atomic levels is larger up to 30 % for H projectiles, 15% for Li projectiles. Ground state relative error is small in range of 1 - 5%.

Relative error for Na is irrelevant.

asztalosors, 2019-09-03:
The regression tests should include more pure plasma oriented benchmarks and be done not in the framework of a Unittest.

closed 2019-09-03 by asztalosors

gergopokol, 2019-09-04:
For the time being, we can accept the regression tests (actually RENATE benchmarks) with the given precisions. Expected precisions are to be updated in the next release.

Later true regression tests are to be set up using old runs of RENATE-OD as reference.