plasma-sds/renate #75: Regression tests are to be set up using old runs of RENATE-OD as reference.

gergopokol, 2019-09-04:
This will need to be done by proper CI tools.

gergopokol, 2019-09-04:
The CI tool handling the data for regression tests could be considered to be used to handle data for unit tests, as well...

asztalosors, 2019-10-07:
Based on conversations with Micskei Zoltan:

1. Cheap and easy method:
Regression testing (SystemTests) would be implemented and should be able to be handled by Travis.
Most likely platform will be a downloadable database elaborated on current: RENATE - ROD acceptance tests. 
Each release of RENATE-OD will add to the database of Test cases, with which current version has to agree fully or to a specific, determined degree.
TestSuite vs TestCase might be more suited. 

2. Not cheap and easy but more eloquent:
More complex methods might need the installation of certain ROD releases and live compare their results based on given input parameters.

gergopokol, 2019-10-08:
Discussion with @asztalosors and @mattngc 
Integration tests with CHERAB are to be executed by running the same test cases as for the regression test, but through CHERAB, and compare the results.

gergopokol, 2019-10-08:
We will definitely go for option 1:
> 1. Cheap and easy method:
>        Regression testing (SystemTests) would be implemented and should be able to be handled by Travis.
>        Most likely platform will be a downloadable database [...]

asztalosors, 2019-10-20:
Point of discussion: The current assert functions benchmark the numerical solver. crm_solver however has 3 more features that it computes:

1. Total beam attenuation.
2. Beam emission of any supported level.
3. Relative beam evolution.

These should be included in the benchmark TestCase? if so then dedicated assert functions need be written to support such features.

asztalosors, 2019-12-10:
Unittest platform is not compatible with a more dynamic testing infrastructure. Exp: same TestCase to be initialized by different beam species; choose from xml content whether a benchmark is expected to fail of pass ...
For further optimization pytest might be explored.

asztalosors, 2019-12-10:
Current version loads pre programmed testcases from "test_dataset\crm_systemtests\actual\.." and runs all benchmarking in one single test function. 
Fails will indicate in error messages, exactly where and which benchmark failed.

Issue to be closed.

closed 2019-12-10 by asztalosors