plasma-sds/renate #78: Travis test update

gergopokol, 2019-09-04:
A branch is to be created from development for the purpose.
Travis is to be configured to run new unit tests and the regression test.

gergopokol, 2019-09-04:
travis-update branch created for the purpose.

asztalosors, 2019-09-04:
All tests have been included in travis and run automatically.  Problems:
1. Tests for atomic db in for the interpolators are failing. Both locally and on travis. 
2. Regression tests fail on travis but OK locally. There seems to be an issue with data fetching from public.
3. Data fetching routines produce and incredible amount of SPAM for user notification.

I consider the task a success as all the tests are running. Investigation of 4 and 1 account of fails is required. !!

gergopokol, 2019-09-05:
Yep. This needs to be fixed before release. Can do in this branch.

Data fetching noise will be solved with improved multi-level logging. Not now.

asztalosors, 2019-09-05:
Related Issue: #79

asztalosors, 2019-09-05:
Related Issue: #80

asztalosors, 2019-09-05:
Issue fixed.

closed 2019-09-05 by asztalosors