plasma-sds/renate #83: Impurity scaling

asztalosors, 2019-09-05:
Mass scaling should improve handling of impurities. Update should be made to the CRM routines.
Temperature profiles of various impurities will  be scaled and shifted with their mass.

asztalosors, 2019-09-06:
Papers describing the handling of multiply charged ion impact excitation and ionization are as follows:
https://iopscience.iop.org/article/10.1088/0953-4075/27/1/017 for Li
https://doi.org/10.1016/j.adt.2008.06.001 for Na

The idea behind the mass scaling of impurities is to adjust the impurity temperature profile to a proton equivalent temperature profile.
Cross-section considering single charged ions such as: H, D, T, He(+), Li(+), .... are treated equivalently. 

Mass scaling of higher charged states is not trivially related to proton masses. Investigation is in order (see papers).

asztalosors, 2019-09-07:
OK. So we are at an impasse.

The charge scaling functions have an embedded mass scaling as well.
q = 1 charge is based on Li + H (A=1)
q = 2 charge is based on Li + He(A=4)
q = 4 charge is based on Li + Be(A=9)
q = 5 charge is based on Li + B(A=11)
q = 6 charge is based on Li + C(A=12)
q = 8 charge is based on Li + O(A=18)
For Sodium:
q = 1 charge is based on Na + H (A=1)
q = 2 charge is based on Na + He(A=4)
q = 4 charge is based on Na + Be(A=9)

All H isotopes and single ionized impurities (He3+), (He4+) .... will have their temperature profiles re-scaled to proton impact energies: T_{D*} = T_{D}*scale; where scale = sqrt(m_{D} / m{H}).
All q = 2 charged particles will be scaled to He impact energies .... and so on.

This means that correct mass scaling requires 2 mass values, one of the species in question, and one bound to the charged state. The latter information is not stored in ROD and needs be.

asztalosors, 2019-09-07:
This leaves us with multiple options:

1.(fastest) Add a dictionary by linking charged states to atomic numbers. This means to append the atomic_db.
2.(preferred) Create a pandas where the actual relative atomic masses are stored for all of the atoms of interest. Add this to atomic_db and do it more correctly.
3. Use via pip isntall external libraries such as or git checkout:

3.1. https://pypi.org/project/mendeleev/ (this one seems nice. might be more problematic on windows but seems cute.)
3.2. https://github.com/jhykes/nuclide-data seems huge and slow to parse through.
- 

-

gergopokol, 2019-09-09:
For the moment 1. (fastest) option is to be implemented, but in a structure that facilitates later more correct treatments. It has been decided that the `coefficientmatrix `module will receive the rates produced specifically to interactions with the plasma species defined in the input profiles. There will be an intermediate layer between `coefficientmatrix` and `atomic_db` that implements the hacking on the data given in `atomic_db` that is necessary to produce the rates for the requested interactions. 

This will facilitate the inclusion of further atomic databases. Accordingly, present atomic_db is to be renamed to `atomic_db_renate`. The intemediate layer is to be called `atomic_db`.

asztalosors, 2019-09-11:
Apparently routines do work properly. 
There might be an error in the input files for ROD. Namely:
H beam in He plasma. ion1: (q,Z,A) = (1,1,2), which is D. Benchmark fails
Li beam in D plasma. ion1: (q,Z,A) = (1,1,2), which is D. Benchmark OK
Na beam in H plasma. ion1: (q,Z,A) = (1,1,2), which is D. Benchmark fails.

asztalosors, 2019-09-11:
The current state of RENATE benchmarks:

1. H plasma in Na beam. OK
[image]

2. D plasma in Li beam. OK
[image]

3. He plasma in H beam. Fail.
[image]
[image]

asztalosors, 2019-09-11:
A detailed study on mass and q scanning is required for comprehensive RENATE-ROD acceptance testing.

asztalosors, 2019-09-11:
Atomic DB unittesting needs to be fixed