plasma-sds/renate #84: Update expected precisions of regression tests.

gergopokol, 2019-09-06:
This should be done after solving the #83 impurity scaling issue.

gergopokol, 2019-09-09:
beamlettest.py is to be renamed crmacceptencetest.py, and the class is to be renamed accordingly. Test cases are to be extended to cover more relevant beam cases (H, D, Li, Na) in pure H plasma.

gergopokol, 2019-09-09:
Related issue: #50

gergopokol, 2019-09-09:
Test cases are to operate with realistic profiles (W7-X island profile in the IAEA benchmark) to avoid numerical uncertainties at the beginning of "constant" profiles. All beam species are to be tested at 100 keV beam energy, and all test cases are to have their own test functions.

SzondyB, 2019-09-11:
Added acceptance tests for standard H plasma and different beam species. RENATE-OD's level naming convention implemented with the use of inv_atomic_dict. Failures due to division by 0 are avoided with if statements. Each of the acceptance tests ran without failure.