plasma-sds/renate #97: Adding Noise to Synthetic signals

asztalosors, 2019-11-04:
A module is needed to add noise to modeled BES signals. The basic description:

1. Read and write synthetic BES signals
2. Add noise.
-------------------------------------------
Noise model should be based on paper of: Dunai: https://doi.org/10.1063/1.3488458 for APD detectors.
1. A Poisson noise should be added based on photon count within a sampling period.
2. Based on the mean photon value of the signal, and a SBR a background photon noise will be added to the signal.
3. An amplifier noise will be added that scales with the photon current strength. See equation (1).
4. The Johnson noise will be added following a Gaussian distribution.

asztalosors, 2020-08-11:
Related issue for data reading #112

andorfi, 2020-08-17:
[first suggestions.pdf](https://github.com/gergopokol/renate-od/files/5083672/first.suggestions.pdf)

In this pdf you can read our first suggestions to the work.

asztalosors, 2020-09-04:
After discussions with @gergopokol the following class structure was agreed upon.
[image]

asztalosors, 2020-09-04:
The class design principles should correlated to: http://butunclebob.com/ArticleS.UncleBob.PrinciplesOfOod

asztalosors, 2021-07-07:
Added MPPC detector modelling #160

gergopokol, 2021-07-09:
Reviewed.

closed 2021-07-09 by gergopokol