plasma-sds/renate #105: Access data

asztalosors, 2019-12-10:
Common features of getdata and putdata should be extracted into a parent class from which the latter two would inherit data access protocols.

gergopokol, 2019-12-11:
Hint for remote file operations:

`ssh -i KEY -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null deep_backup@beam.rmki.kfki.hu "cp -la /data/scratch/deep_backup/current /data/scratch/deep_backup/$date"`

gergopokol, 2020-02-03:
Should write check_private_server_data_path(self) in accessdata.py

gergopokol, 2020-02-08:
Paramiko http://www.paramiko.org/ could have been an option, but it is not in Anaconda, and thus may not be installed at the sites of deployment, so its use is to be avoided.

Simpler solution is to use the os package, like:
os.system('ssh -i key.ppk deep.ppk -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null pokol@deep.reak.bme.hu "ls"')
(Still having problems with unprotected private key in Windows...)

gergopokol, 2020-02-11:
This should be able to return the list:
`subprocess.check_output('ssh -i key.ppk -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null pokol@deep.reak.bme.hu ls path')`

Still having a problem with the keyfile. Now it is "invalid format"...

asztalosors, 2020-08-18:
With regard to recent developments regarding the use of venv in environments such as IMAS; paramiko should be reconsidered. As RenateOD is supposed to work with CHERAB in linux, adding paramiko to the requirements and install procedure should not be a difficult task. This might solve our problem of server data access and functions with urllib using private data keys.

asztalosors, 2020-09-24:
Class works as expected. Development on the class is finished.

closed 2020-09-24 by asztalosors