plasma-sds/renate #110: Scipy constants instead of burnt-it variables

matyasaradi, 2020-07-25:
Use [scipy constants](https://docs.scipy.org/doc/scipy/reference/constants.html) in utility/constants.py instead of burnt-it variables!

asztalosors, 2020-08-17:
You are right. Though if one looks up the physical  constants (the definition for constants used here is slightly expanded) available within scipy: https://docs.scipy.org/doc/scipy/reference/constants.html is quite wast. There are certain constant values available in packages like (https://pypi.org/project/mendeleev/) which are not available in scipy.

The usage of a constants object or collection native to RenateOD should provide a selfconsisten collection of constants. For the stand alone uses data populated by scipy and augmented with discrepancies. For the use of RenateOD in IMAS or other integrated coding platform, constant db-s provided by said platforms should be adapted by the before mentioned object/collection.

asztalosors, 2020-09-05:
Not surprising, the switch to different constants broke some test. The Unit tests require update.

asztalosors, 2021-07-08:
This issue is fixed and should be considered closed for the moment.

closed 2021-07-08 by asztalosors