plasma-sds/renate #111: Pandas, h5 error

andorfi, 2020-08-10:
Installed for the first time, ran at development branch, and I got the following error:

C:\Anaconda\python.exe "C:\Program Files\JetBrains\PyCharm Community Edition 2020.2\plugins\python-ce\helpers\pydev\pydevconsole.py" --mode=client --port=65256
import sys; print('Python %s on %s' % (sys.version, sys.platform))
sys.path.extend(['C:\\Users\\István\\Documents\\RenateOD', 'C:/Users/István/Documents/RenateOD'])
Python 3.7.6 (default, Jan  8 2020, 20:23:39) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.12.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 7.12.0
Python 3.7.6 (default, Jan  8 2020, 20:23:39) [MSC v.1916 64 bit (AMD64)] on win32
from crm_solver.beamlet import Beamlet
Backend Qt5Agg is interactive backend. Turning interactive mode on.
b = Beamlet()
Data is NOT located in the common local directory: C:\Users\István\Documents\RenateOD\utility\..\common_data\beamlet/testimp0001.xml
Data is NOT present in the user local directory: C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/testimp0001.xml
Attempting to download from server: data@deep.reak.bme.hu:~/private_html/renate-od/beamlet/testimp0001.xml
testimp0001.xml           | 0 kB |   0.3 kB/s | ETA: 00:00:00 | 100%
Data has been downloaded from the server to the user local directory: C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/testimp0001.xml
ElementTree read from: C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/testimp0001.xml
Beamlet.param read from file: beamlet/testimp0001.xml
Data is NOT located in the common local directory: C:\Users\István\Documents\RenateOD\utility\..\common_data\beamlet/test_impurity.h5
Data is NOT present in the user local directory: C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5
Attempting to download from server: data@deep.reak.bme.hu:~/private_html/renate-od/beamlet/test_impurity.h5
test_impurity.h5          | 23 kB |  23.8 kB/s | ETA: 00:00:00 | 100%
Data has been downloaded from the server to the user local directory: C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5
Traceback (most recent call last):
  File "C:\Anaconda\lib\site-packages\pandas\io\pytables.py", line 655, in open
    self._handle = tables.open_file(self._path, self._mode, **kwargs)
  File "C:\Anaconda\lib\site-packages\tables\file.py", line 315, in open_file
    return File(filename, mode, title, root_uep, filters, **kwargs)
  File "C:\Anaconda\lib\site-packages\tables\file.py", line 778, in __init__
    self._g_new(filename, mode, **params)
  File "tables/hdf5extension.pyx", line 492, in tables.hdf5extension.File._g_new
tables.exceptions.HDF5ExtError: HDF5 error back trace
  File "C:\ci\hdf5_1545244154871\work\src\H5F.c", line 509, in H5Fopen
    unable to open file
  File "C:\ci\hdf5_1545244154871\work\src\H5Fint.c", line 1400, in H5F__open
    unable to open file
  File "C:\ci\hdf5_1545244154871\work\src\H5Fint.c", line 1546, in H5F_open
    unable to open file: time = Mon Aug 10 16:26:39 2020
, name = 'C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5', tent_flags = 0
  File "C:\ci\hdf5_1545244154871\work\src\H5FD.c", line 734, in H5FD_open
    open failed
  File "C:\ci\hdf5_1545244154871\work\src\H5FDsec2.c", line 346, in H5FD_sec2_open
    unable to open file: name = 'C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0
End of HDF5 error back trace
Unable to open/create file 'C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Anaconda\lib\site-packages\IPython\core\interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-3-7f2c74c930d5>", line 1, in <module>
    b = Beamlet()
  File "C:\Users\István\Documents\RenateOD\crm_solver\beamlet.py", line 20, in __init__
    self.__read_beamlet_profiles()
  File "C:\Users\István\Documents\RenateOD\crm_solver\beamlet.py", line 35, in __read_beamlet_profiles
    self.components = utility.getdata.GetData(data_path_name=hdf5_path, data_key=['components']).data
  File "C:\Users\István\Documents\RenateOD\utility\getdata.py", line 45, in __init__
    self.read_data()
  File "C:\Users\István\Documents\RenateOD\utility\getdata.py", line 78, in read_data
    self.read_h5_to_pandas()
  File "C:\Users\István\Documents\RenateOD\utility\getdata.py", line 100, in read_h5_to_pandas
    self.data = pandas.read_hdf(self.access_path, key=self.data_key[0])
  File "C:\Anaconda\lib\site-packages\pandas\io\pytables.py", line 397, in read_hdf
    store = HDFStore(path_or_buf, mode=mode, errors=errors, **kwargs)
  File "C:\Anaconda\lib\site-packages\pandas\io\pytables.py", line 537, in __init__
    self.open(mode=mode, **kwargs)
  File "C:\Anaconda\lib\site-packages\pandas\io\pytables.py", line 687, in open
    raise IOError(str(err))
OSError: HDF5 error back trace
  File "C:\ci\hdf5_1545244154871\work\src\H5F.c", line 509, in H5Fopen
    unable to open file
  File "C:\ci\hdf5_1545244154871\work\src\H5Fint.c", line 1400, in H5F__open
    unable to open file
  File "C:\ci\hdf5_1545244154871\work\src\H5Fint.c", line 1546, in H5F_open
    unable to open file: time = Mon Aug 10 16:26:39 2020
, name = 'C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5', tent_flags = 0
  File "C:\ci\hdf5_1545244154871\work\src\H5FD.c", line 734, in H5FD_open
    open failed
  File "C:\ci\hdf5_1545244154871\work\src\H5FDsec2.c", line 346, in H5FD_sec2_open
    unable to open file: name = 'C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0
End of HDF5 error back trace
Unable to open/create file 'C:\Users\István\Documents\RenateOD\utility\..\data\beamlet/test_impurity.h5'

asztalosors, 2020-08-10:
Attempted to recreate error. Works just fine with my system. Seems to be a version issue.
Items to be checked:
python version:

- python 3.7.6 vs python 3.6.4

And the pandas versions of the anaconda distributions.
My version of pandas: 0.22.0 please @andorfi check your pandas version with following commands in console: >>> import pandas <<< followed by >>> pandas.__version__ <<<

asztalosors, 2020-08-10:
The error propagates from: self.data = pandas.read_hdf(self.access_path, key=self.data_key[0]), through a lengthy chain returning: unable to open file: name = 'C:\Users\István\Documents\RenateOD\utility..\data\beamlet/test_impurity.h5', errno = 2, error message = '**No such file or directory**', flags = 0, o_flags = 0. 
I have gone through the deprecation in pandas from 0.20.1 to the current versions (https://pandas.pydata.org/docs/whatsnew/index.html) and .read_hdf has suffered no significant deprivations that should result in the before mentioned errors. The current calling sequence is still recommended (https://pandas.pydata.org/docs/user_guide/io.html#hdf5-pytables). 

Based on the error the exact location and existence of the .h5 file has to be verified (though it is there and doubtful to be the source of the error). The path also contains "á" charcters, which should be excluded ASAP as the source of the error.

asztalosors, 2020-08-10:
**Managed to reproduce error:** 

Python 3.6.4 |Anaconda, Inc.| (default, Jan 16 2018, 10:22:32) [MSC v.1900 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 6.2.1
Python 3.6.4 |Anaconda, Inc.| (default, Jan 16 2018, 10:22:32) [MSC v.1900 64 bit (AMD64)] on win32
from crm_solver.beamlet import Beamlet
Backend Qt5Agg is interactive backend. Turning interactive mode on.
C:\Program Files\JetBrains\PyCharm Community Edition 2017.3.3\helpers\pydev\_pydev_bundle\pydev_import_hook.py:20: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  module = self._system_import(name, *args, **kwargs)
b=Beamlet()
Data is NOT located in the common local directory: H:\Codes\renáte-od\utility\..\common_data\beamlet/testimp0001.xml
Data is located in the user local directory: H:\Codes\renáte-od\utility\..\data\beamlet/testimp0001.xml
ElementTree read from: H:\Codes\renáte-od\utility\..\data\beamlet/testimp0001.xml
Beamlet.param read from file: beamlet/testimp0001.xml
Data is NOT located in the common local directory: H:\Codes\renáte-od\utility\..\common_data\beamlet/test_impurity.h5
Data is located in the user local directory: H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5
Traceback (most recent call last):
  File "D:\Software\Anaconda\lib\site-packages\pandas\io\pytables.py", line 580, in open
    self._handle = tables.open_file(self._path, self._mode, **kwargs)
  File "D:\Software\Anaconda\lib\site-packages\tables\file.py", line 320, in open_file
    return File(filename, mode, title, root_uep, filters, **kwargs)
  File "D:\Software\Anaconda\lib\site-packages\tables\file.py", line 784, in __init__
    self._g_new(filename, mode, **params)
  File "tables\hdf5extension.pyx", line 487, in tables.hdf5extension.File._g_new
tables.exceptions.HDF5ExtError: HDF5 error back trace
  File "C:\ci\hdf5_1510862531115\work\src\H5F.c", line 586, in H5Fopen
    unable to open file
  File "C:\ci\hdf5_1510862531115\work\src\H5Fint.c", line 1236, in H5F_open
    unable to open file: time = Mon Aug 10 20:12:10 2020
, name = 'H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5', tent_flags = 0
  File "C:\ci\hdf5_1510862531115\work\src\H5FD.c", line 809, in H5FD_open
    open failed
  File "C:\ci\hdf5_1510862531115\work\src\H5FDsec2.c", line 346, in H5FD_sec2_open
    unable to open file: name = 'H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0
End of HDF5 error back trace
Unable to open/create file 'H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "D:\Software\Anaconda\lib\site-packages\IPython\core\interactiveshell.py", line 2910, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-3-6a21cc1a0686>", line 1, in <module>
    b=Beamlet()
  File "H:\Codes\renáte-od\crm_solver\beamlet.py", line 20, in __init__
    self.__read_beamlet_profiles()
  File "H:\Codes\renáte-od\crm_solver\beamlet.py", line 35, in __read_beamlet_profiles
    self.components = utility.getdata.GetData(data_path_name=hdf5_path, data_key=['components']).data
  File "H:\Codes\renáte-od\utility\getdata.py", line 45, in __init__
    self.read_data()
  File "H:\Codes\renáte-od\utility\getdata.py", line 78, in read_data
    self.read_h5_to_pandas()
  File "H:\Codes\renáte-od\utility\getdata.py", line 100, in read_h5_to_pandas
    self.data = pandas.read_hdf(self.access_path, key=self.data_key[0])
  File "D:\Software\Anaconda\lib\site-packages\pandas\io\pytables.py", line 349, in read_hdf
    store = HDFStore(path_or_buf, mode=mode, **kwargs)
  File "D:\Software\Anaconda\lib\site-packages\pandas\io\pytables.py", line 467, in __init__
    self.open(mode=mode, **kwargs)
  File "D:\Software\Anaconda\lib\site-packages\pandas\io\pytables.py", line 612, in open
    raise IOError(str(e))
OSError: HDF5 error back trace
  File "C:\ci\hdf5_1510862531115\work\src\H5F.c", line 586, in H5Fopen
    unable to open file
  File "C:\ci\hdf5_1510862531115\work\src\H5Fint.c", line 1236, in H5F_open
    unable to open file: time = Mon Aug 10 20:12:10 2020
, name = 'H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5', tent_flags = 0
  File "C:\ci\hdf5_1510862531115\work\src\H5FD.c", line 809, in H5FD_open
    open failed
  File "C:\ci\hdf5_1510862531115\work\src\H5FDsec2.c", line 346, in H5FD_sec2_open
    unable to open file: name = 'H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0
End of HDF5 error back trace
Unable to open/create file 'H:\Codes\renáte-od\utility\..\data\beamlet/test_impurity.h5'

**Summary:** Paths MAY NOT HAVE any special characters !!!!

closed 2020-08-10 by asztalosors