plasma-sds/renate #112: Synthetic data reader

andorfi, 2020-08-10:
The first step for making the noise modeling for RENATE OD is requires data access scripts. A personaliization of the data routines of utility package is required.
This code is needed to able to read to a synthetic signal class. Date reading should occur within the __init()__ of the class.
This class is able to get data from an external source (most likely Pandas format), if Pandas is not available, external data should be loaded from H5 or text.

asztalosors, 2020-08-12:
The GetData object can not load and transform array type of data from txt file to an array format. Might need an expansion of the GetData object.

andorfi, 2020-08-12:
[first suggestions.pdf](https://github.com/gergopokol/renate-od/files/5065558/first.suggestions.pdf)
In this pdf you can read our first suggestions to the work.

asztalosors, 2021-07-08:
The synthetic data reader class is an ongoing development. Its scope is met within the noise modelling branch. For further development, a new issue would be recommended.

closed 2021-07-08 by asztalosors