plasma-sds/renate #120: Noise generator class and related unittesting.

asztalosors, 2020-09-04:
As stated in issue #119 the random number generator class RandomState will be considered parent class to the Noise generator as is has already all built-in features for random number generation.

The class needs be able to generate a set of various noise types and random numbers. Signal values will not be stored within the object.
Noise types to be included: 
1. Photonic noise generator (shot noise generator)
2. Dark noise generator
3. Johnson noise generator
4. Voltage noise generator

Unittesting of Noise module required. Related issues: #118 
Parent issue: #97

asztalosors, 2020-10-16:
@andorfi If basic physics within a class is changed. The corresponding test cases have to updated to avoid failure!!

gergopokol, 2021-02-22:
.txt files to be cleaned up.

asztalosors, 2021-07-08:
Class is finished and tested.

closed 2021-07-08 by asztalosors