plasma-sds/renate #122: PMT Noise generator class

asztalosors, 2020-09-04:
This task is related to #97 noise modelling and build on the developed Noise class featured in #120, which is a parent class to PMT.
Using the noise generation routines available in Noise. The PMT class will create an accurate noisy signal version based on ph/s input data.

Unittesting of the PMT class is related to issue:

gergopokol, 2021-02-22:
Poisson routine is to be amended to calculate Poisson for low expected value and Gauss for high expected value.

asztalosors, 2021-04-09:
Minor corrections are required by the code:
* The emitted electron population should consist of: expected electron generation + background electron generation + dark electron generation. --> this population should be magnified by the PM tube.

asztalosors, 2021-04-10:
To asses Poisson noise generation vs Gaussian noise generation, The routines also require a PMT Gaussian noise generation routine  packadge. The Hamamatsu PMT brochure provides adequate descri[tion of the process in question.

asztalosors, 2021-04-10:
On the note of Drak noise generation: The Dark Noise generation by a Gaussian distribution is not correct in current approach where the statistical electron propagation through the PMT is modelled. 

The AUG PMT (R928) features a Dark Noise value range of 3 - 50nA and a Gain factor of 1E7 (~6^9). Take the value of 25nA = 25E-9A as dark current. Accounting for the Coulomb charge (1.6E-19) results in 15.62 E10 electrons/s. Account for the Gain and the dark cathode electron emission amounts to 1.56 E4 electron/s resulting in 0.07 dark electrons in a measurememtn time frame. This is can not be approached with Poisson realistically. 

* One idea would be to denerate a Dark electron suppliment of Poisson(1.5E4) and hope one goes above 2E5 (samling frequency). This does not work, however from a case of 7 million samples the max value was 1.7E4, which of course is not ralistic.
* A better approach would be to apply a uniform distibution and if the random number is in the 7% (1.5E4 / 2E5) then a dark electron is created and added to the total electron count.

If there is a desire to experiment with higher dark current values, such as: 1.6 microA or higher would ensure that on average every timestep has 1 dark electron (for AUG @ 2E5). For such a case a switch can be applied to use a poisson distribution to modulate them if we want to. 

I would go with door nr. 2.

asztalosors, 2021-07-05:
https://www.hamamatsu.com/resources/pdf/etd/PMT_handbook_v3aE.pdf

andorfi, 2021-07-05:
After the question for APD and PMT dark noise modeling, I've found the following read: https://www.hamamatsu.com/resources/pdf/ssd/e03_handbook_si_apd_mppc.pdf, where the APD's noise relations written at page 4. By this, there are two types of dark signal at an APD, the amplified dark current (Idg) and the so called surface leakage current, whic isn't multiplied (Ids). If we define the scanned data as the sum of them, the earlier used expression (used by J. Hadobas et al.) is correct. And respect to the earlier commented PMT handbook, we need to use a different implementation for APD and PMT's dark noise.

asztalosors, 2021-07-07:
Minor changes are still required in dark noise generation. Dark Noise generator should have a net_gain input in the setup routine. This should be set to 1 for APD or maybe set as default =1. Except for PMT where it is 2*gain*noise_gain.

asztalosors, 2021-07-07:
Done and ready. The class contains a basic Gaussian and a Detailed noise model.

closed 2021-07-07 by asztalosors