plasma-sds/renate #137: Unittesting and cleanup of the conver.py functions

gergopokol, 2020-09-28:
Some functions here could be substituted with CHERAB methods. Others need unittesting.

gergopokol, 2026-09-18:
We should avoid CHERAB dependency...

asztalosors, 2026-10-01:
There are two functions: convert_beamlet_profiles and convert_beamlet_profiles_to_si. Neither are used and serve no discernable role.

- convert_beamlet_profiles -- converts RENATE.h5 output into RENATE OD h5, however renate-od already reads RENATE.h5 files.
- convert_beamlet_profiles_to_si -- converts density values into si density by multiplying this one col. with 1e19

Recommend removing the functions @gergopokol

gergopokol, 2026-10-08:
> There are two functions: convert_beamlet_profiles and convert_beamlet_profiles_to_si. Neither are used and serve no discernable role.
>
> * convert_beamlet_profiles -- converts RENATE.h5 output into [...]

These routines were supporting the generation of inputs. These have been obsoleted by adding the utility/Input.py routines in #200 .
Can be removed.

asztalosors, 2026-10-08:
Done

closed 2026-10-08 by asztalosors