plasma-sds/renate #144: Data Purge in Beamlet

asztalosors, 2020-10-05:
In direct relation to issues: #142 and #141 A further development of the Beamlet class is required.
It is needed to remove all computations from a beamlet object.

asztalosors, 2020-10-06:
There seems to be a hiccup in the labeling H-D-T atomic levels.  As the code removes data by level name Li and Na is easy to handle as they are named: '2s', '2p' ...etc. The H-D-T levels however are label quite ubiquitously '1', '2' etc. making finding these level names amongst the labels quite difficult as other labels fall in the mix as well such as: 'ion1', 'ion2', ... etc

I propose two recommendations: 
* All transitions should be marked by 'trans' such as: '2s-2p' --> 'trans 2s-2p' or '1-2' --> 'trans 1-2'
* Return to a better H level naming convention: '1' --> '1n' or '1-2' --> '1n-2n'

gergopokol, 2020-10-06:
I would go for '3n-->2n' and '2p-->2s'.

asztalosors, 2020-10-06:
Regression and Acceptance tests Fail. Which is to be expected as profile atomic labels differ !!

asztalosors, 2020-10-06:
Modifications to code successful and all test run OK.

closed 2020-10-06 by asztalosors