plasma-sds/renate #158: Dependency update

gergopokol, 2021-02-22:
An automatic Pull Request had been generated by [Dependabot](https://docs.github.com/en/github/managing-security-vulnerabilities/configuring-dependabot-security-updates) caused by a security alert: https://github.com/gergopokol/renate-od/pull/156
This proposed to update the required lxls package version. 

The Travis runs to an error with the old version, so the change is not optional, but the new version seems to cause an error in some of the tests. A branch was created from development for the purpose of resolving this and a manual merge was applied to merge the Depemdabot pull request: [dependency-update](https://github.com/gergopokol/renate-od/tree/dependency-update)

All dependency versions are to be updated.

gergopokol, 2021-06-22:
Got a new one: now lxml should be updated to 4.6.3. This automation was switched off, and we will handle this manually. (Alerts still enabled.)

asztalosors, 2021-06-22:
Problem fixed.

closed 2021-06-22 by asztalosors