plasma-sds/renate #177: RENATE-OD relative path issue

Balazs-Peter, 2021-08-05:
ROD doesn't seem to be able to use a relative path input when creating a beamlet. On the other hand, it works when I give the absolute path.

**Code:**
from crm_solver.beamlet import Beamlet

#path to xml file relative to the script:
data_path='ROD_inputs/xml/basic_N200_a0.2_paH_baLi_10keV.xml'

b = Beamlet(param=None, profiles=None, components=None, solver='numerical', data_path=data_path)

**Error message:**
FileNotFoundError: The requested file: ROD_inputs/xml/basic_N200_a0.2_paH_baLi_10keV.xml does not exist! Check input!

gergopokol, 2021-08-06:
Full output of code is requested for debugging.

asztalosors, 2022-02-09:
So: RenateOD has a well defined and expected data file system.

The user may set up any location desired (user_local_data_directory>data</user_local_data_directory>). 
- If an absolute path is given ("D:/test") then that will be considered the user data file and all in and output will be expected within that structure
- If relative path is given then the user data folder is expected to be within the file struct of renate-od.

All atomic and utility data will be stored there. Currently, renate does not support loading XML data from a different location than that which is defined by the user data space.

If we want to have the option of providing an XML with a proper absolute path, that can be arranged.

asztalosors, 2022-02-10:
The agreement is as follows: 
data_path_name should be evaluated whether it is an absolute path or not. 

Based on that decision, GetData should act accordingly.

asztalosors, 2022-02-10:
Done

closed 2022-02-10 by asztalosors