plasma-sds/renate #221: Interdependency of packages is to be reviewed and corrected is needed.

gergopokol, 2022-02-07:
Packages are supposed to have a level of independence in deployability.

mferenczy, 2026-03-26:
Can I get an example for this from the codebase? Is it about the utility code that other projects would also want to use or something else?

asztalosors, 2026-10-06:
The above commit fixed the issue raised in the Istvan branch. The coefficient.py needs a lot more work. See #259

asztalosors, 2026-10-06:
To limit and reduce interdependence, I would suggest the following changes to the files object and functions:
- Once all codes regarding refactoring are done and final, the code should be restructured so that the utility holds all code that other submodules use commonly. This requires a more in-depth discussion at that point.
- Remove all testing functions to a /test folder outside of the core code.
- Remove all redundant  / unfinished / stale modules: cherab_demos, renate, imas_wrapper, imas_utility from core code.
- The above two points are linked to pip installability of renate
- This issue is also somehow part of this bundle #76

gergopokol, 2026-10-08:
> To limit and reduce interdependence, I would suggest the following changes to the files object and functions:
>
> * Once all codes regarding refactoring are done and final, the code should be [...]

OK.
> * Remove all testing functions to a /test folder outside of the core code.

OK.
> * Remove all redundant  / unfinished / stale modules: cherab_demos, renate, imas_wrapper, imas_utility from core code.

OK, move to a dedicated subdirectory that is not src. This way, we will have the code but it will not be installed by pip.
> * This issue is also somehow part of this bundle [The module structure is to be changed just before release. All executable sub-modules should be included in the main renate-od module. #76](https://github.com/plasma-sds/renate/issues/76)

#76 is actually a subissue of this one.

Balazs-Peter, 2026-10-08:
When structural changes of this kind will be on the table, I think it will be worth reconsidering the setup and import convention too. As of now, submodules like "utility" are visible for the kernel, making imports like `import utility.convert as uc` possible. This makes it seem like utility is an independent package, which is very unconventional. I imagine it could confuse someone too when debugging. Something like `import renate_od.utility.convert as uc` would be more obvious. Should I make an issue for this?

asztalosors, 2026-10-08:
You are right in your view. For now lets not. I would suggest that a separate branch be made for this issue from dev. once the Release 1.4.0 Cylce has come to an end and this to be the final task. For now, I would suggest to merge refactor into dev. and solve as many open issue as are. If possible not to clutter the Project layout only when necessary.

asztalosors, 2026-10-08:
Once dev was updated with state of the art refactor, smaller task branches should be made to address issue clusters and smaller tasks towards relsease 1.4.0