plasma-sds/renate #248: Bug in H (n=1) + e --> H (n=2) + e cross-section

asztalosors, 2022-07-05:
There seems to be an issue with the H + e electron excitation cross-section.

Formula # 10 in line 58 of hydrogenic_cross_sections.py shows the cross-section being described by a single function, whereas in the reference of Janev on p6 of https://inis.iaea.org/collection/NCLCollectionStore/_Public/25/024/25024274.pdf?r=1 there are 3 functions and energy ranges.

Clarification required.

Balazs-Peter, 2026-09-18:
The first energy range is 10.2 - 11.56 eV, the second is 11.56 - 12.23 eV, and the third is above 12.23 eV. The implemented formula is for the third one of course, as the other ranges seemed insignificant. Would a comment in the code stating that the third range is used resolve the issue?

gergopokol, 2026-09-18:
> The first energy range is 10.2 - 11.56 eV, the second is 11.56 - 12.23 eV, and the third is above 12.23 eV. The implemented formula is for the third one of course, as the other ranges seemed insignificant. Would a comment in the code stating that the third range is used resolve the issue?

Correct asymptotic behaviour is to be checked!

Balazs-Peter, 2026-09-19:
Low energy behavior was conformed to be incorrect. See figures, where the blue curve is produced by RENATE-OD, and the black one is the original.

[image]

Balazs-Peter, 2026-09-19:
I stiched together the first and third energy region in the formula with the help of tanh weighing functions, since the middle energy region is just a single value in the same vicinity. The resulting cross-section is very close to the original, see figure below.

[image]

closed 2026-09-30 by asztalosors