plasma-sds/renate #254: AtomicDB to HDF writer

Balazs-Peter, 2022-07-10:
Create a class to save/read an AtomicDB instance to/from an HDF file.

asztalosors, 2026-09-30:
Yes

asztalosors, 2026-09-30:
#263 Is being continued and merged into this issue.

Balazs-Peter, 2026-10-01:
Found an earlier writer that was hard-coded for hydrogen beams in atomic.tools. I am going to replace it with a general writer based on AtomicDB.

asztalosors, 2026-10-01:
Good there are two feature to be added
- [ ] create a unittest: toolstest.py where the features are automatcally tested.
- [ ] the conversion function should have a better name (guess it does m2 to cm2) and moved to the utility.convert.py as thatone already fearures a convert_from_cm2_to_m2 fucntion

asztalosors, 2026-10-02:
Great success. I would also add a loading test as well to make a full cyrcle test. Saves and stores into .h5, loads it and make sure all expected data is in it. After that the issue should be done

Balazs-Peter, 2026-10-05:
I would load in the saved HDF file and create a new AtomicDB instance from it, or even run a beam evolution calculation with it. If you agree, I'll go forward with this, but I'm not sure how to instantiate an AtomicDB with an arbitrary file path to an HDF file. The "rate_type" argument looks for a folder in "data", as far as I can tell. What do you recommend?

asztalosors, 2026-10-06:
Indeed that seems the way to go. Considering the current state of implementation the following would be true to the best of my understanding and as I read through the code:
- place the new h5 in a folder next to deafult - the name of which should be rate type. For now that is the form to load data. ex: internal .... or such
- the bottleneck is the __set_impurity_mass_scaling_dictionary(self) routine that requires impurity rates to be present up to charge 11 (A,Z) = (11, 23), effectively Sodium.
- To test the write load features as RENATE type data: I recommend featuring a large dataset with all reates required: electron, proton .... up to charge 11.
- The test might be simple enough as to load data from renate default--> save as internal --> load as renate internal. 
- Other means would be to first self.impurity_impact_trans = self.get_from_renate_atomic('impurity_transition') and then build __set_impurity_mass_scaling_dictionary(self) , from its size but information is missing from h5, what the target projectiles are. This step would be preferred but should be part of the next dev. cycle. Go with the option from above.

Balazs-Peter, 2026-10-07:
Since the AtomicDB test is done with dummy data, I make a folder for a "writer_test" rate type like this:

`os.mkdir(os.path.join(os.path.dirname(__file__), '..', 'data', 'dummy', 'atomic_data', 'dummy', 'rates', 'writer_test'))`

This is where the output file is written. Let me know if there is a better way to get to the appropriate location.

In the test of the output file, an AtomicDB instance with the same components as the original test instance of AtomicDB is created from the file, and beam evolution calculation is run with it. Is this okay?

asztalosors, 2026-10-07:
I would not merge multiple tests into one. I would have more and smaller test to be able to more easily locate the point of failure. 
- One test to check if the writing works. --> shows save features.
- One test to see whether a DB can be built from saved file. --> running the beam evo shows the DB works and is comaptible and build occurs.
- Test a few random Temp values for a few interpolators between original and the saved+stored db versions. --> shows that the stored values and OK.

Balazs-Peter, 2026-10-07:
If I understand correctly, you propose 3 tests. Since tearDown is called after each, this would look like this:

1. Write out
2. Write out, read in, beam evolution
3. Write out, read in, compare values

If there's a problem with writing out, all three still fails, and in general, this doesn't look like the best practice. After consulting our friend Claude, one option is to write out and read in a file in a setUpClass method, that is called only once, and not deleting them in tearDown. Then, beam evo and interpolation values can be tested individually, while the file and read-in variable is only deleted in tearDownClass. Otherwise, it also suggests to make one test for the whole workflow. Let me know what you think.

asztalosors, 2026-10-07:
Yes, the overall setupclass and teardownclass seems the way to go. I differ to your judgement. The aim is to be able to have a resolution in the points of failure.

asztalosors, 2026-10-08:
Cool. Seems like a done deal.

closed 2026-10-08 by asztalosors